nasa/T-MATS #58: calc_Pstatic_TMATS.c problem

Jackeygou2015, 2015-12-29:
Because of the different mass flow in subsonic and supersonic with identical (Pt,Tt,far,Aexit) input, the guess values x1=0.2 & x2=0.5 just are suitable for subsonic flow to calculate Ps, which will cause error in supersonic flow. I think x1&x2 initial values should be distinguished.  Of course, the secant method also easily cause this problem when faced to mass flow curve like this.

GRC-RHC-Sim, 2015-12-30:
What values are you having issues with? I'm not sure what you are referring to as x1 and x2.

This block is meant to back calculate the static conditions based on total conditions. You can assume either a MN or an Area (it says A throat, but is is more just an area. it is not trying to act like a nozzle). If you know the area it then uses the MN as an initial guess. Because there are two answers for each Pt (a subsonic and supersonic) if you guess below 1 you will get a subsonic answer and if you guess above 1 you get supersonic.  This is was it was meant to do anyhow.

If you have specific values that cause the block to crash feel free to post them and I can take a look. 

Thanks,
-Jeff

Jackeygou2015, 2015-12-30:
When PtIn & TtIn are defined, the same Area in subsonic or supersonic can get the same Wout, but  corresponding to different Ps. The calc_Pstatic_TMATS() function calculate Ps based on MN iteration. If the initial values is (x1=0.2,x2=0.5), the MN iteration will stop and get Ps at subsonic MN, but the fact is that the Wout&Area may be refered to the Ps at supersonic MN,  so the Ps answer is not correct. 
the code as followed,
 /*----- use these 2 points to start the solution ----*/
/*   where x ==> MN, y ==> W, and z ==> Ps */
x1 = 0.2; 
x2 = 0.5;

as well as the x1&x2 is not input value in this function.  

Thanks.

chapman178, 2016-01-13:
That’s true, so it is up to the user to know if they are dealing with subsonic or supersonic and fill out the initial MN accordingly.

-Jeff

GRC-RHC-Sim, 2016-02-29:
Closing because the question was answered. If there is still concern please post again.

closed 2016-02-29 by GRC-RHC-Sim