nasa/T-MATS #62: Some issues about Linearize block

ck1313, 2016-03-03:
In the help of Linearization Library Block, the dimension of N_XPer_M should be rxm, the dimension of N_UPer_M should be rxn.
In the MATLAB Function, line26-28, why transpose BIn and x_dot_hist,  when length(x_dot) is 1? 
if length(x_dot) == 1
    B = BIn';
    x_dot_log = x_dot_hist';
else
    B = BIn;
    x_dot_log = x_dot_hist;
end

GRC-RHC-Sim, 2016-03-03:
ck1313,  

Thanks for finding this.  The matrix dimensions have been updated in 3e319dfd8705cee337c8aadcca489cb623952b07   as of 3/3/2016.

As for the second questions.  This code is a product of MATLAB doing strange things when talking between MATLAB functions and Simulink.  x_dot_log and x_dot_hist are saved x_dot baseline and perturbed values.  These are exported to a matlab memory block outside of the MATLAB function (perhaps I should have used the memory feature of this block... but I didn't).  Anyhow,  if x_dot is length 1, then saving the history of them is a vector and for some reason Simulink will flip them when it exits and enters a MATLAB function. So, when the old x_dot_hist is brought in to be updated it must be transposed so the rest of the code will work. Note: an actual matrix will be read correctly.

So, the delay block in Simulink is set to a  1xm,  the MATLAB function interprets that to be an mx1, the code then flips it (back to 1xm) so that all the sizes workout in the code, at the end of the code the vector is flipped again to a mx1 so that when it exits the MATLAB function block it will return to being an 1xm.   I may have the row vs col thing mixed up, but hopefully this makes it understandable.  

I would probably consider this a MATLAB/Simulink bug, and if it gets fixed in the future the code will need to be updated.

closed 2016-03-03 by GRC-RHC-Sim

ck1313, 2016-03-04:
Thanks for your guidance.

ck1313, 2016-03-04:
I have another question.
Line 68-79:
% log x_dot and y values
if LenPV == 1
    if i == b;
        x_dot_log(1) = x_dot;
        y_log(1) = y;
    end
else
    if i > a && i<= LenPV + a+1;
        x_dot_log(:, i - a) = x_dot;
        y_log(:, i - a) = y;
    end
end
When LenPV=1(only one  disturbance variable) and i=2, x_dot_log(1) saved the baseline values. But no code to save the perturbed values. So we can not get the ABCD.