nasa/T-MATS #64: Another issue about Linearize block

ck1313, 2016-03-06:
I have another question about Linearize block.
In the MATLAB Function, line 68-79:
% log x_dot and y values
if LenPV == 1
if i == b;
x_dot_log(1) = x_dot;
y_log(1) = y;
end
else
if i > a && i<= LenPV + a+1;
x_dot_log(:, i - a) = x_dot;
y_log(:, i - a) = y;
end
end
When LenPV=1(only one disturbance variable) and i=2, x_dot_log(1) saved the baseline values. But no code to save the perturbed values. So we can not get the ABCD, when LenPV=1.

GRC-RHC-Sim, 2016-03-07:
Good question.   I played around with the block a bit and couldn't even figure out how this LenPV could be set to 1...  The mask requires an input for X pert and U pert,  and LenPV is the combined length of those two.  I think this logic was placed in anticipation of an issue, but the issue looks like it sorts itself out and this code isn't used. 

I did a few runs with the block to determine how it would react to a situation where the U pert was set to a number, but u was not being used to generate x_dot (ie.  the single value case) and it returns B = 0, which makes sense.  I also put in a 0 for U pert size and it returned a nan.  I prefer the zero, so I would suggest placing a U pert value even if you have no U input.

If you have any specific issues with the Linearization block (situations where it fails) let me know so I can take a look.

I will leave this item open as a reminder to go back and clean up this section of code. Also, a statement in the block help should be added to discuss this situation.

ck1313, 2016-03-07:
Put in a 0 for U pert size and it returned a nan.
I think  it is because line 103-104:
A = ABavg(:,1:LenX);
B = ABavg(:,LenX+1 :LenXU);
LenX = LenXU. We do not have (LenX+1 :LenXU), so the index bigger than dimensions. B is nan.
The same to D.

GRC-RHC-Sim, 2016-03-08:
Is that not acceptable?
If a user is putting a 0 in for pert sizes they most likely made a mistake, so sending out a Nan seems ok to alert them that all is not well, though maybe a separate message would be more obvious.

GRC-RHC-Sim, 2016-05-20:
closing this conversation out.  If there is concern the issue was not addressed well enough and an error message should be added, feel free to open it back up.

Thanks,
-Jeff

closed 2016-05-20 by GRC-RHC-Sim