nasa/T-MATS #72: Gear ratio in the gearbox block

Haroldinho, 2016-07-05:
I was using the gearbox block to model a turboprop.
When converting the torque and rpm from a free power turbine output to a propeller input, 
it seems that both the torque and the rpm are being divided by the gear ratio in that block.
I believe that the torque should be multiplied by the gear ratio, whereas the rotation rate should be divided by the gear ratio, in order for energy to be conserved across that block.

GRC-RHC-Sim, 2016-07-06:
Dividing or multiplying by the gear ratio is a matter of frame of reference. I think the confusion for this is coming  from how the block is created. It was created to work directly with the compressor and turbine blocks, so it has a virtual side A and side B. This does not line up with inputs and outputs, because the compressor block needs speed as an input and outputs a required torque. This means the block must input side A speed (from shaft) and side B torque (from component) , and output side A torque (sent to the shaft) and side B speed (sent to the component). 

This leads to the following relationship:
Gear Ratio = NmechA/NmechB = TrqB/TrqA
Which, lines up with what you are saying (depending on frame of reference)

Note: when building code, this block will be broken up into two sections. (for example with a compressor.  the speed conversion will run before the compressor block, then the torque portion will run after the compressor.

closed 2016-07-06 by GRC-RHC-Sim

Haroldinho, 2016-07-06:
Thank you for the clarification. 
It makes a lot more sense now.