nasa/T-MATS #85: the VSV of High Pressure Compressor

chengdingding, 2017-12-12:
Hello, teacher. I am a Ph.D. student and  very interested in T-MATS.  So I intend to use T-MATS as a research platform to study the multi-variable control of engine. And we are going to use VSV as one of the input variables, but TMATS's VSV doesn't seem to be work. Can you tell me how to do it or would you like to share the map about Alpha of JT9D High Pressure CompressorVG?

GRC-RHC-Sim, 2017-12-13:
Hello Chendingding, 

Thanks for taking an interest in T-MATS. Generally it is expected that a T-MATS user will make use of 2d maps that assumes VSV position is ideal during operation.  Off nominal VSV can be taken into account in a few ways with T-MATS. 
1) the CompressorVG block may be used with a 3d map, with the third dimension typically being VSV angle or off-nominal angel depending on how the maps are constructed. As you mention T-MATS does not come with a 3d map, so you would need to generate that yourself or find an example from literature.  
2) A 2d map can also be used by adjusting the map scalars to reflect off nominal VSV position.  Again, how much to adjust would be left up to you.  I don't know how realistic they need to be, but you may be able to piece together simple representative values by looking through texts on gas turbine performance and the effect of VSV.

-Jeff

chengdingding, 2017-12-14:
Dear Dr.Jeff        
                                                                                                                                     
Thank you very much for your reply and the explanation of patience.          
                                                                                                                             
As you said, We want to properly adjust 2D map as a 3Dmap and the specific adjustment as follows:   
                                                                     
1.we set up the Compressor Map Flow Array is ‘cat(3,MWS.HPC.WcArray,MWS.HPC.WcArray*1.1)’;
2.the Compressor Map Pressure Ratio Array is‘ cat(3,MWS.HPC.PRArray,MWS.HPC.PRArray*1)’;
3.and Compressor Map Efficiency Array is ‘cat(3,MWS.HPC.EffArray,MWS.HPC.EffArray*1.05)’ .
4.At the same time, other parameters do not change.        
                                                                       
Do you think it is reasonable for us to set up this? Can you give us some advice?   

Once again, thank you!

-chengdingding

chengdingding, 2017-12-14:
Dear Dr.Jeff

Thank you very much for your reply and the explanation of patience.

As you said, we want to properly adjust 2D map as a 3Dmap and the specific adjustment as follows:

1.we set up the Compressor Map Flow Array is ‘cat(3,MWS.HPC.WcArray,MWS.HPC.WcArray*1.1)’;
2.the Compressor Map Pressure Ratio Array is‘ cat(3,MWS.HPC.PRArray,MWS.HPC.PRArray*1)’;
3.and Compressor Map Efficiency Array is ‘cat(3,MWS.HPC.EffArray,MWS.HPC.EffArray*1.05)’ .
4.At the same time, other parameters do not change.

Do you think it is reasonable for us to set up this? Can you give us some advice?

Once again, thank you!

-chengdingding

chengdingding, 2017-12-14:
Dear Dr.Jeff       
                                                                                                                                      
Thank you very much for your reply and the explanation of patience.         
                                                                                                                                
As you said, we want to properly adjust 2D map as a 3Dmap and the specific adjustment as follows:  

1.We set up the Compressor Map Flow Array is ‘cat(3,MWS.HPC.WcArray,MWS.HPC.WcArray*1.1)’;
2.The Compressor Map Pressure Ratio Array is‘ cat(3,MWS.HPC.PRArray,MWS.HPC.PRArray*1)’;
3.And Compressor Map Efficiency Array is ‘cat(3,MWS.HPC.EffArray,MWS.HPC.EffArray*1.05)’ ;
4.At the same time, other parameters do not change.       

Do you think it is reasonable for us to set up this? Can you give us some advice?     

Once again, thank you!

-chengdingding

chengdingding, 2017-12-14:
Dear Dr.Jeff

Thank you very much for your reply and the explanation of patience.

As you said, we want to properly adjust 2D map as a 3Dmap and the specific adjustment as follows:

1.We set up the Compressor Map Flow Array is ‘cat(3,MWS.HPC.WcArray,1.1*MWS.HPC.WcArray)’;
2.The Compressor Map Pressure Ratio Array is‘ cat(3,MWS.HPC.PRArray,1*MWS.HPC.PRArray)’;
3.And Compressor Map Efficiency Array is ‘cat(3,MWS.HPC.EffArray,1.05*MWS.HPC.EffArray)’ ;
4.At the same time, other parameters do not change.

Do you think it is reasonable for us to set up this? Can you give us some advice?

Once again, thank you!

-chengdingding

chengdingding, 2017-12-14:
Dear Dr.Jeff

Thank you very much for your reply and the explanation of patience.

As you said, we want to properly adjust 2D map as a 3D map and the specific adjustment as follows:

1.We set up the Compressor Map Flow Array is ‘cat(3,MWS.HPC.WcArray,1.1*MWS.HPC.WcArray)’;
2.The Compressor Map Pressure Ratio Array is‘ cat(3,MWS.HPC.PRArray,1*MWS.HPC.PRArray)’;
3.And Compressor Map Efficiency Array is ‘cat(3,MWS.HPC.EffArray,1.05*MWS.HPC.EffArray)’ ;
4.At the same time, other parameters do not change.

Do you think it is reasonable for us to set up this? Can you give us some advice?

Once again, thank you!

-chengdingding

chengdingding, 2017-12-14:
Dear Dr.Jeff

Thank you very much for your reply and the explanation of patience.

As you said, we want to properly adjust 2D map as a 3D map and the specific adjustment as follows:

1.The Compressor Map Flow Array is ‘cat(3,MWS.HPC.WcArray,MWS.HPC.WcArray*1.1)’;
2.The Compressor Map Pressure Ratio Array is‘ cat(3,MWS.HPC.PRArray,MWS.HPC.PRArray*1)’;
3.And Compressor Map Efficiency Array is ‘cat(3,MWS.HPC.EffArray,MWS.HPC.EffArray*1.05)’ ;
4.At the same time, other parameters do not change.

Do you think it is reasonable for us to set up this? Can you give us some advice?

Once again, thank you!

-chengdingding

GRC-RHC-Sim, 2017-12-14:
I can't comment on the scalar values, but the method for generating the 3d map looks good.

You would probably want the default map to represent the ideal VSV position, so alpha would represent off-nominal position.  If this is the case I would suggest having at least 2 scalars, for positive and negative off-nominal VSV angle: 
‘cat(3, MapArray*-Y, MapArray, MapArray*X)’;
X and Y values determined empirically. 

-Jeff

chengdingding, 2017-12-15:
Thank you very much for your advice, and I think it really makes sense. 
I hope to have a chance to consult you in the future.Thank you again for your help. 

-chengdingding

closed 2018-01-03 by GRC-RHC-Sim