nasa/T-MATS #89: formatting of license

thorade, 2018-08-14:
Very minor improvement: If the license is in a certain format, it will be recognized & displayed by github.
https://github.com/blog/2252-license-now-displayed-on-repository-overview
https://help.github.com/articles/adding-a-license-to-a-repository/

The easiest way to have the right format is to delete the license file and then re-add it from the github web interface by clicking the "Create new file" button and then specifying as filename `License.txt` or `LICENSE.md`. A dropdown menu will appear and ask what license template to use.

chapman178, 2018-08-14:
Will do.

Thanks,
-Jeff

https://github.com/blog/2252-license-now-displayed-on-repository-overview
https://help.github.com/articles/adding-a-license-to-a-repository/

chapman178, 2018-08-22:
Completed

closed 2018-08-22 by chapman178