nasa/T-MATS #95: Opspec/Linearize

davidvanwyk, 2019-10-21:
I think adding the ability to make calls to linearize in MATLAB to linearize a T-MATS model in Simulink would be quite beneficial. 

Is this already possible or has anyone implemented this? Otherwise I can take a look and see if I can come up with a solution for people to use.

bluequaza, 2019-10-23:
I currently use T-MATS and got some results. 
- If i use linearize block in T-MATS library, i get ABCD vectors, that differ from matlab linearize function
- Results is sensitive with initial condition and convergence problem too
- If i use frequency estimation model (my custom code) in matlab to estimate frequency response (bode plot) i got another result from multi-sine input signal
So now i have 3 different results with 3 different frequency response. And experimental results show that third method is more appropriate.

davidvanwyk, 2019-10-23:
Initial condition dependence is expected due to gas turbines being highly non-linear as a function of their state (quite well contained by spool speed, altitude, mach number and dT). 

The process which I currently use is to create a small perturbation around the operating point I want the linearization around (established separately using a steady state model and known ambient condition + mechanical speed) and then use system identification to pull the linear model from the results. This is tedious and requires an external toolbox, however, so I'd like to avoid it if possible.

I'd probably look into taking a snapshot of the model to get opspec data, but one might also be able to use operspec and just remove the solver block from the loop and get MATLAB to solve for the input states to try and force the flow errors to 0... I'd have to investigate though. 

One would then attempt to linearize around that point. I'd then compare this to the linearization achieved from the sysid approach and ensure cross-validity there. I'm not sure how well the linearize function will play with the do-while structure of T-MATS, though, so it might not be possible and perhaps a more custom approach would need to be developed.

chapman178, 2019-10-23:
I have not used the Linearize block from a matlab script.  Though I would think the sim command should work for this purpose. 

Good comments on the linearization.  It is also important to remember that the state space representation of a system is not unique. So it is possible that each method is generating ABCD matrices that look different, but operate the same.

davidvanwyk, 2019-10-23:
@chapman178 - seconded RE state space representation not being unique. However, if he's getting frequency-domain responses that differ notably using the various methods then something is likely amiss.

Regarding linearizing from MATLAB. It becomes quite important when one wants to do gain-scheduled controller designs, because working directly in Simulink for this purpose becomes rather unwieldy and automating the designs from MATLAB becomes a lot more attractive.

Presumably you've not tried the linearize command (https://www.mathworks.com/help/slcontrol/ug/linearize.html) with the current do-while architecture? I'm wondering if it would struggle and just produce something not physically meaningful by not allowing flow matching to be achieved.

chapman178, 2019-10-23:
You know,  when developing that block, I looked at the linearize command first and don't recall exactly why we didn't use it.  

Maybe it was in the controls toolbox and we wanted T-MATS to work w/o any extra toolboxes,  or perhaps (like you said) the do... while doesn't work well with it.  Or maybe we just wanted more direct control over what it was doing.  

Whatever the reason, it would be an interesting to try it out and see if it works with a simplified simulation.

davidvanwyk, 2019-10-23:
Certainly might be interesting. I'll give it a look on my side when I find a moment and report back too. 

What would the current recommendation be for automated linearization? It seems you allude to a very similar process to what I've been doing in the AGTF30 repo. How is it done there?

bluequaza, 2019-10-23:
I have an example and hope it will answer the question about linearize block in T-MATS. 
Example is about single spool turbojet. Because T-MATS only uses one differential equation with shaft speed. Mathematically, it will result in first order transfer function.
 If rotational speed is output. We know that H(s)=CB/(sI−A)+D. State space has different values of ABCD matrix, but in this case, CB/A = DC gain, A is pole location, so only CB varies, and D=0. But Linearize block in T-MATS results D matrix isnt zero.

bluequaza, 2019-10-23:
Complex block structures inside engine model, and many with discontinuities may affect linearize results  of linearize matlab function. By default, linearize in matlab is exact linearize, individually linearizes each block in your Simulink model and combines the results to produce the linearization of the specified system. Maybe mathematical pertubation method is alternative option i will try.

chapman178, 2019-10-23:
Blue, 

Yes, that answers the question.  Also reminded me of why the block was implemented in the way it was.  Essentially, In this problem we know we want one state per shaft. It was found that the linearize function would adjust the number of states automatically to the system and sometimes it was found that the calculated number didn't line up with the number we knew.  

As for the D matrix thing. Yes, I would think D should be 0 in this case. The method works by perturbing each state and watching the outputs. U (if U is fuel only) would not directly effect the speed so should be zero. There may be an issue with the problem setup.  My first guess would be that you have a state in there somewhere. The linearization block needs to be run on a steady-state system, so it needs to be in either a do .. while loop,  or a simulink model that has no states. It could be that there is a state on the shaft and the linearize block is iterating with the active integrator. Just a guess,  w/o looking at your setup. If i get some time I will take a look at the block see if i can re-create your issue.  

Thanks,
-Jeff

davidvanwyk, 2019-10-23:
The number of states isn't really too much of a physically meaningful parameter for the most part in the MATLAB/Simulink way of doing things (this should be fairly obvious given how prolific functions like minreal() are) and are essentially just a means to an end to generate correct time/frequency domain models. 

I'll have a look at the method you've used - I don't think I've actually used your linearize functionality before. Perhaps I was using an older T-MATS that didn't include it (and have stuck with it because I transitioned everything to SI units... speaking of which. Can I add a pull request that adds this feature to your block masks?). 

I do think linearize _might work and warrants investigation. But if your custom method is easy to automate then I don't really see any problems going that route.

Interestingly, if one looks at something like y = [N, EGT], u = Wf, you do see some feedthrough for the EGT because fuel flow essentially results in an instantaneous rise in temperature from the combustor outlet. This isn't physically true (there's usually a slight delay), but is pretty close to true. EGT also exhibits behaviour that certainly cannot be captured with a single state model. 

So while the models might only explicitly have a single state (shaft speed), they implicitly can be higher order due to the solver and flow matching impacts. Just something to keep in mind.

closed 2024-01-03 by chapman178