nasa/T-MATS #109: Cantera v1.0 no longer available

ThomasKenningsNASA, 2024-05-20:
T-MATS integration with Cantera states that Cantera v1.0 should be used. However, v1.0 is not available for download from Cantera's website or GitHub. Where can Cantera v1.0 be downloaded from? Or will other available versions of Cantera work?

I've tried using the JT9D_SS_Cantera example with Cantera v3.0.0 installed, but the function "importPhase" has apparently been removed from Cantera and replaced by the "Solution" function. Once I replaced that function call, the model still has some strange periodic behavior. I'm suspecting that if I had Cantera v1.0 then this issue wouldn't appear.

ThomasKenningsNASA, 2024-05-20:
As an update: I tried uninstalling Cantera v3.0 and installed v2.0 instead. This allowed the "importPhase" function to work as was written in the JT9D example. However, strange periodic spikes still occur in the solutions. 

Feel free to close this issue, although do note that the version of Cantera recommended by the T-MATS documentation cannot be found on the internet.

chapman178, 2024-05-20:
Thanks for bringing this up! I recently got a refresh and was trying to get this set back up and had some issues.  I have not had a chance to troubleshoot them yet. We will look into this and see if we can put out a fix.

closed 2024-08-09 by chapman178